SJSURoboticsTeam/urc-electrical-2023 #2: RMD USBC Breakout

swiftrax, 2022-10-22:


swiftrax, 2022-10-30:
Homing pin connected to magnet switch. Other end of magnet switch is GND. Similar to this 
[image]

a3has, 2022-11-03:
This is Revision A of the usb c breakout for the motors, please review it and double check our work. Feedback is greatly appreciated. 
There are some issues in this board we would love to get help with:
1- we were not able to find the correct phoenix headers for the homing pin and ground on easyEDA pro, we used a placeholder until it can be replaced (it unfortunately does not have 3D view also).
2- I think out vias are too small for the power lines, we could make them wide if needed.
-Aboudi & Dawinder
[ProProject_USB C BREAKOUT_2022-11-03.zip](https://github.com/SJSURoboticsTeam/urc-electrical-2023/files/9933761/ProProject_USB.C.BREAKOUT_2022-11-03.zip)

a3has, 2022-11-18:
Board all done. 
TVS added
zero DRC errors
all traces are correct width.

Ready for final review before order.

swiftrax, 2022-11-20:
good work!

closed 2022-11-20 by swiftrax