SJSURoboticsTeam/urc-electrical-2023 #8: USB C testing

swiftrax, 2022-10-30:
Test USB C breakouts to drive a motor and find out the lengths of cables we need going from body > leg and from rotunda > arm. 

USB C to C cable should be rated about 240W for legs maybe 100W for arm. Test and log results

kammce, 2022-10-30:
Maybe test that the USB C cables can 

1. Handle a steady 48V voltage without overheating. Such overheating would be due to some part internal to the cable breaking down and now shorting to gnd.
2. Handle a steady 5A of current @ whatever voltage you can muster. 30V, which I believe is the limit of our power supplies in 141 should be fine

Then we can attempt to test against the motors. Always put the motors last in terms of testing. Let's not break more motors like we did last year.

kammce, 2022-10-31:
We should standardize for 240W for everything. This way, if it's overkill for the arm, we get more room for adding stuff like additional cameras and such.

a3has, 2023-01-03:
Cable has been tested to withstand 5A with no detrimental heat on cable and usb c breakout. Thermal camera was used to record the highest temperature of 30 degrees C. Khalil approved the use of the 1m - 3.3Ft cable tested. This exact cable is of suitable length to use from power board to backpack with enough slack. More testing needs to be done with a load tester but not necessary. Will note down any worthwhile details when testing with backpack.

closed 2023-07-27 by swiftrax