SJSURoboticsTeam/urc-electrical-2023 #43: Make kill switch PCB

swiftrax, 2023-08-06:
Make a PCB that is for the kill switch to make it smaller using the following

- [ ] [Remote Switch](https://a.co/d/1UJ9zEK)
- [ ] [Push Switch](https://www.amazon.com/APIELE-Emergency-Latching-Stianless-Mouting/dp/B08QD31NNC/)
- [ ] [Inrush Thermistor](https://www.amazon.com/Aqua-Rite-Thermistor-2R025-Current-Limiter/dp/B071HX7XNY/ref=sr_1_3?keywords=inrush+current+limiter&qid=1691282520&sprefix=inrush+%2Caps%2C124&sr=8-3)

swiftrax, 2023-08-06:
Ordered 2 of the parts listed above

kammce, 2023-08-06:
Just a note that the inrush thermistor I found and listed in the first comment should work for our needs, but its quite expensive from Amazon. _Might_ be able to find some cheaper on digikey or mouser. I tried looking for them in LCSC but couldn't find them. Maybe they go under a different name on LCSC? I also cannot find any SMD versions of this, so through hole will definitely be the way to go as far as I can tell.

kammce, 2023-08-06:
As for the relay, I wonder if there is a way to eliminate the need for the buck converter on this? 🤔 Just a thought, if such a scheme is too complicated then we can stick with the buck converters. Maybe having another inductor in series can add some resistance drop the voltage at the coil? But I'm not sure if that results in the more or less power consumption.

swiftrax, 2023-08-06:
> As for the relay, I wonder if there is a way to eliminate the need for the buck converter on this? 🤔 Just a thought, if such a scheme is too complicated then we can stick with the buck converters. [...]

The remote switch runs off 3.3V - 12V so I think we have to keep the buck on there

kammce, 2023-08-06:
> The remote switch runs off 3.3V - 12V so I think we have to keep the buck on there

Great point! Yeah, buck stays then :)

swiftrax, 2023-08-06:
> Just a note that the inrush thermistor I found and listed in the first comment should work for our needs, but its quite expensive from Amazon. _Might_ be able to find some cheaper on digikey or [...]

Couldn't find SMD versions of these either, was able to find a cheaper version the SL32 but from what I've read its weaker than the AS32. Price is about the same on Amazon vs Digi ($7 Amazon / $7 Digi) but Amazon is cheaper because shipping.

swiftrax, 2023-08-06:
[Analog Switch to lower power consumption](https://www.analog.com/en/technical-articles/analog-switch-lowers-relay-power-consumption.html#:~:text=On%20the%20other%20hand%2C%20the,lower%20voltage%20(Figure%201).)
We can do something like this and throw a 3.3V regulator since the drop out voltage is around 1.2V.

kammce, 2023-08-06:
Thats not a bad idea 😄 We just need to verify that a steady 3v3 on the coil will actuate it well.

swiftrax, 2023-08-10:
There's two ways we can do this:
- Figure out relay holding current
- Switch voltages using the 3V3 Regulator
[image]

swiftrax, 2023-08-13:
Components and circuit verified. Schematic made pending review then going to layout and route pcb to mechanical's specification. (@MayaEnriquez )

DolphinGui, 2024-02-10:
This is now the stop sign kill switch PCB.

closed 2024-02-10 by DolphinGui