SJSURoboticsTeam/urc-electrical-2023 #47: ESP32 Camera

swiftrax, 2023-08-06:
Make board to interface with ESP32 Sense board and a BP70A battery
[Xiao ESP32 Sense](https://www.aliexpress.us/item/2255799825490363.html)

swiftrax, 2023-08-07:
@MayaEnriquez we might need help with coming up with a case / PCB for this project

swiftrax, 2023-08-13:
[image]

Connector [LCSC P0.5 24Pin](https://www.lcsc.com/product-detail/_Shenzhen-Kinghelm-Elec-_C2797213.html)

swiftrax, 2023-08-13:
[Possible Mounting Solution](https://a.co/d/2T1jpuD)
[Quick Release Mount](https://a.co/d/iARgRxp)

swiftrax, 2023-08-13:
Need to find a place for the antenna

DolphinGui, 2024-02-10:
The ESP-cam is nearly complete now, and due to time constraints utilizes an COTS esp board modified with a battery connector.

closed 2024-02-10 by DolphinGui